Hilti-Research/hilti-slam-challenge-2022 #1: Changelog

hilti-challenge, 2022-03-14:
Subscribe to this issue to get updated when/if we modify the problem statement. Please DO NOT add comments here, create a new issue instead.

mcamurri, 2022-05-04:
### 04/05/2022

- Updated the URDF and the frame diagram as discussed in #4

hitdshu, 2022-05-16:
Just want to know if the deadline could be extended for one day...

I come across this competition last week and I want to submit my results. But downloading the whole dataset costs me a lot of time....

hilti-challenge, 2022-05-16:
Hi hitdshu, thank you for your interest in participating the challenge. We have just extended the deadline by one day (until May 17!

hitdshu, 2022-05-16:
You guys are so kind and nice. Thanks very much.

YoshuaNava, 2022-05-16:
@hilti-challenge The website still shows the previous submission deadline:

[image]

hitdshu, 2022-05-17:
@hilti-challenge Well, looks like the results are already evaluated for all the submitted system....

https://submit.hilti-challenge.com/

hitdshu, 2022-05-17:
I'm still hoping to submit the results today. However, not allowing it is understandable.

hilti-challenge, 2022-05-17:
> @hilti-challenge The website still shows the previous submission deadline:
> 
> [image]

we will still accept submissions until end of today!

hitdshu, 2022-05-17:
Thanks very much. I've submitted my results. Thanks for your kindness.