Hilti-Research/hilti-slam-challenge-2022 #2: Where to find the calibration file of Hilti SLAM Dataset 2021?

oywlNKU, 2022-03-30:
Dear developers, thanks for your amazing work!

I'm trying to use the dataset to test my algorithm, but I didn't find the link to download the calibration file of Hilti SLAM Dataset 2021. Can you please share the calibration file of Hilti SLAM Dataset 2021? Thank you very much!

Best regards.

hilti-challenge, 2022-03-30:
Hi,
we provide the raw [calibration data](https://storage.googleapis.com/hsc2022/calib_03_2022-03-02-11-27-22.bag) + [calibration board description](https://storage.googleapis.com/hsc2022/april_grid_2500x1500_7x12_15cm.yaml) for camera calibration as well as the Kalibr results for the [individual cameras]. (https://storage.googleapis.com/hsc2022/calibration/2022322_calibration_files.zip) (with respect to the imu). The zip file also contains the extrinsic parameters between lidar and imu.

Alternatively, you can also use the [CAD ](https://storage.googleapis.com/hsc2022/cad/Phasma25.stl) as an initial guess.

Hope you find this information useful.

oywlNKU, 2022-03-30:
Thank you for your reply.
But I noticed that the file you provided  is the calibration file of Hilti SLAM Dataset 2022, may I have the calibration file of Hilti SLAM Dataset 2021?

oywlNKU, 2022-03-30:
Wow, I have seen that. Thank you very much!

closed 2022-03-30 by hilti-challenge