Hilti-Research/hilti-slam-challenge-2022 #5: Need a dataset to analyze the IMU performance

JokerJohn, 2022-04-06:
Hello, we have checked the internal and external parameters calibration results of imu and lidar provided by the official website. The IMU part only provides the three-axis bias of the accelerometer and gyroscope calibrated with Kalibr. But we also want to know the white noise of the accelerometer and gyroscope. Could you provide the corresponding data set? For example, Collect the data while the IMU is Stationary, with a two or three hours duration. These noises of the IMU are critical to the performance of our algorithm. Thank you!

hilti-challenge, 2022-04-06:
Yes, we will shortly upload such a dataset.

hilti-challenge, 2022-04-26:
we have uploaded a 1.5 hour [stationary IMU dataset ](https://storage.googleapis.com/hsc2022/calibration/2022-04-25-12-52-59-stationary-imu.bag).

closed 2022-04-26 by hilti-challenge