Hilti-Research/hilti-slam-challenge-2022 #6: Error submission

hellovuong, 2022-04-08:
Hello, thank you for orginize this challenge. I run into error when trying to submit. After submit zip file and report. The score appear to have error as following:

hellovuong, 2022-04-11:
Thank you, looking forward to the update.

hilti-challenge, 2022-04-26:
the automatic evaluation system should be functional right now.

closed 2022-04-26 by hilti-challenge

IamPhytan, 2022-04-26:
On the same topic, I tried to submit my results and I got the following error :

> Traceback (most recent call last):   File "/var/task/evaluation-2022/batch_evaluation.py", line 224, in <module>     evaluate_submission(submission_path, groundtruth_path)   File "/var/task/evaluation-2022/batch_evaluation.py", line 87, in evaluate_submission     ape_stats, dense_trajectory, traj_est_aligned, traj_ref = evaluate(est_file, ref_file,submission_path)   File "/var/task/evaluation-2022/evaluation.py", line 240, in evaluate     umeyama_parameters = traj_est_aligned.align(traj_ref_sync, correct_scale=False, correct_only_scale=False)   File "/var/task/evo/core/trajectory.py", line 203, in align     r_a, t_a, s = geometry.umeyama_alignment(self.positions_xyz.T,   File "/var/task/evo/core/geometry.py", line 70, in umeyama_alignment     raise GeometryException("Degenerate covariance rank, " evo.core.geometry.GeometryException: Degenerate covariance rank, Umeyama alignment is not possible

Is this an error on the server ? Or is our data not coherent with what is accepted by the server ?

hellovuong, 2022-04-26:
Hi @IamPhytan I tried to submit this afternoon, everything works well. I think you should check the file formats. Be sure it is followed the submit instructions

IamPhytan, 2022-04-26:
Yes, I found out it's not an error related to the website, but most probably an error in my trajectories